TcMenu/tcMenuLib #19: Ability to edit text items on device using the rotary encoder / emulator.

davetcc, 2019-05-06:
In order to configure wifi or other more complex settings on the device, a simple system for editing complex text items needs to be implemented.

The probable best way is to have a cursor that is moved between characters with the select key. Once a character is selected for editing, moving the encoder will change the letter underneath the cursor.

dsap4004, 2019-05-11:
I could see a version of this for numerical or time input as well. At the moment the only way to input multiple numbers is multiple menu items.

davetcc, 2019-05-11:
Agreed, we could probably add new menu items that supported times (and possibly eventually dates).

My plan is to make adding new menu item types much easier too. At the moment there's quite a bit of work to do in both the remote API and in the rendering and editing.

Dates are more tricky as there are many rules, however, times have far fewer rules and could be targeted first.

davetcc, 2019-05-24:
Linked with #23

closed 2019-06-23 by davetcc