TcMenu/tcMenuLib #23: Provide a time menu item that can be edited on device.

davetcc, 2019-05-11:
There should be a time menu item that can be edited on the device. The time should be stored as a 32-bit integer, the format of which is yet to be determined.

It should support:

* Formatting to the nearest minute: HH:MM
* Formatting to the nearest second: HH:MM:SS
* Formatting to the nearest millisecond HH:MM:SS.sss

Editing should start with the hour part, and then cycle through the rest of the time parts until all are edited.

dsap4004, 2019-05-11:
For time input you might also want to be able to set 12 and 24 hour mode with an AM PM indicator if let's say you are interfacing with an RTC like a DS3231 or similar.

dsap4004, 2019-05-11:
Also you may want to consider the clock set menu item to be the display as well. You click to set and you can fiddle with the numbers. You click it again to save and you can display what the RTC is set as. When you write the RTC data update to the clock menu item you don't want a callback to fire either. Maybe you are making a stopwatch type application and you need to update the display quickly to see the time counting down. You would be able to set the start time, save it, and the hook other menu items or other buttons into the start and stop capability while displaying the countdown on the same time set menu item.

davetcc, 2019-05-24:
In addition to all the above cases, the date item formatting should allow very basic per locale drawing. The verification and rendering should be pluggable so that a user can easily write an extension for their locale should it not be supported.

The MVP will target Gregorian calendars only with very simple leap year handling.

closed 2019-08-11 by davetcc