TcMenu/tcMenuLib #54: Publish library on Platformio

moritz89, 2020-04-22:
This is by far the most advanced (C++ support, GUI code generator, IO Abstraction) library for generating menus for the Arduino platform. However it lacks visibility and versioning that is given by the Platformio semantic versioning.

Has there been a discussion on publishing the library on Platformio and in the Arduino library manager?

davetcc, 2020-04-22:
Many thanks for the compliment.

Yes, PlatformIO is next on the list once the new designer is released, we painted ourselves into a corner with the old designer UI where it was getting difficult to release it, even with just the Arduino libraries, let alone new platforms and different libraries. However, this is much easier with the new designer.

The new designer will be able to scale to platform IO, and we'll start with Arduino and mbed board support.

davetcc, 2020-04-22:
linked with #49, for which the work will start shortly.

Further, step1 of this will be in the coming days, tcMenuLib will be packaged and put into the Arduino library manager, at this point, it should be available on PIO by default (as is IoAbstraction). See https://platformio.org/lib/show/5738/IoAbstraction

davetcc, 2020-05-05:
Update: we've requested the library to be added to Arduino Library Manager, Platform IO tracks these libraries and adds them to their index too. It will hopefully be there very soon.

See: https://github.com/arduino/Arduino/issues/10139

davetcc, 2020-05-11:
Closing out, library is now listed in Arduino library manager, it should be available very soon (if not already) in PlatformIO, as I think PIO tracks the library manager.

closed 2020-05-11 by davetcc