TcMenu/tcMenuLib #57: Removing selection of + - of large numbers

20163771, 2020-05-18:
I have a situation, where I have to set a frequency. However + - choice at the beginning complicates things as there should not be possibility to put in negative frequency. 

I have tried to delete all of the necessary code from EditableLargeNumberMenuItem.cpp it did remove the + -, however there is only a possibility of inputting 0 or 1 to that position.

I bet it is not the hardest thing to code out, however I am not very skilled and maybe someone else in the future would benefit from it aswell.

davetcc, 2020-05-18:
Yes I can see that this would be generally useful for others. I will look at the best way to handle this. Probably isNegativeAllowed flag or something similar. I’ll add it to the next milestone which probably isn’t that far off.

20163771, 2020-05-18:
Thanks for response. 

Also it would be nice if the dp would not show up when the precision is set to 0.

It just the small things. However, overall this library is a lifesaver. Keep the good work.

davetcc, 2020-08-29:
This has taken a bit longer than usual to complete, we were in the middle of a large scale refactor of the application when the was raised. However, not only is it possible to prevent negative numbers, it should now be fix some problems found at the same time around integer only representation. Will be released soon.

closed 2020-09-01 by davetcc