TcMenu/tcMenuLib #65: “tcMenuDesigner” is damaged and can’t be opened - MacOS

shed909, 2020-07-10:
I'm running MacOS Catalina 10.15.5.

Trying to open tcMenuDesigner 1.4.1 and 1.4.2 gives me:
“tcMenuDesigner” is damaged and can’t be opened. You should move it to the Bin.

1.3.5 works fine, so I'm just using this for now, but just thought I'd report it.

Happy to do any tests.

davetcc, 2020-07-10:
This is caused by it not being Notarised by Apple. Please see https://github.com/davetcc/tcMenu/issues/85 for more information and how to get around it for now.

We are now testing a version of the designer that should be able to go into the Mac App Store. Hopefully, it's now only weeks away.

shed909, 2020-07-13:
Thank you @davetcc .

The workaround posted on the forums works, I just needed to 'sudo su' before I could read items in the Applications folder.

davetcc, 2020-08-18:
Application is now available in the Mac App Store, "tcMenu Designer":

https://apps.apple.com/gb/app/tcmenu-designer/id1527782002?mt=12

I'd be glad to get any feedback on it. It's a native Cocoa app that's based on the same generator as the Windows store generator; which as been around for a while.

davetcc, 2020-08-19:
MacOS version store version has been user-tested and we know it to be working. Closing issue.

closed 2020-08-19 by davetcc