TcMenu/tcMenuLib #72: Compile error on PlatformIO / STM32F103C8

Stanelec, 2020-09-08:
The following error occurs when compiling in PlatformIO for the board bluepill_f103c8 (STM32) :

`
.pio/libdeps/bluepill_f103c8/tcMenu/src/EditableLargeNumberMenuItem.cpp: In member function 'void LargeFixedNumber::convertToBcdPacked(int32_t, int, int)':
.pio/libdeps/bluepill_f103c8/tcMenu/src/EditableLargeNumberMenuItem.cpp:34:41: error: no matching function for call to 'min(int, int32_t)'
   34 |   setDigit(i, std::min(9, value / modulo));
`

The error occurs because the type of parameters passed to min() are different, due to automatic typecasting.

A solution is to explicitly tell the compiler the desired type, as follows:
```c++
		setDigit(i, min<int>(9, value / modulo));
```

Thank you for this nice lib!

davetcc, 2020-09-15:
Apologies it took a while to get to this one. We changed the value recently to uint32_t and should also have changed the modulo. Usually we catch that with either the mbed or ESP toolchain, which are quite fussy about this. Somehow it compiled on both of those platforms!

It would be interesting to know which toolchain you are using with that board. For the STM boards we usually use mbed directly.

davetcc, 2020-09-15:
Please let me know if the change works for you. If it does I'll release 1.6.1

davetcc, 2020-09-16:
I'm going to close this now, if the change has not fixed the issue, please re-open.

closed 2020-09-16 by davetcc

Stanelec, 2020-09-18:
The error occurred with Arduino framework for the bluepill STM32 board. I didn't tried mbed.

Anyway, release 1.6.1 fixes the issue, thank you!