TcMenu/tcMenuLib #84: Error in example "unoSsd1306Ascii", file ssd1306asciiRenderer.cpp

chirrindul, 2021-01-28:
Environment PlatformIO, with added libraries from the IDE. Fragment of platformio.ini
lib_deps = 
	davetcc/tcMenu@^1.7.1
	davetcc/IoAbstraction@^1.6.7
	davetcc/TaskManagerIO@^1.1.0
	davetcc/LiquidCrystalIO@^1.4.0
	greiman/SSD1306Ascii@^1.3.0

New project created and copied the example "unoSsd1306Ascii" to the selected root folder of the project. When compiling this error is raised:

src/ssd1306asciiRenderer.cpp: In member function 'void SSD1306AsciiRenderer::renderList(uint8_t)':
src/ssd1306asciiRenderer.cpp:50:86: error: no matching function for call to 'min(int, uint8_t)'

I don't know if a good solution is to change min for _min, but it resolves the issue.

davetcc, 2021-01-29:
The way to use an OLED display on such a large board is with U8G2 library, and there is an inbuilt plugin for it. Library URL is https://github.com/olikraus/u8g2 but better installed from the Arduino / platformIO library manager.

Even if we fix this you probably can't use this library with ESP32 as it's built for AVR. We just provide the glue code, the library is here https://github.com/greiman/SSD1306Ascii. We've never considered using this lib on anything other than an Uno.

Task: Remove this plugin from all ARDUINO32 boards, it should not have been there.

chirrindul, 2021-01-29:
Thanks for the instructional feedback. Sorry for making you lose your time.

closed 2021-01-29 by chirrindul