TcMenu/tcMenuLib #88: [BETA] Multi edit runtime editor needed for touch screens

davetcc, 2021-02-08:
We need an editor for multi-edit runtime items such as large-number, text, date, time, and IP-address.

Although we can look into better editors over time, initially this is for a menu item dialog with an analog slider representing the range. It will have a next and back button. Depends on #87

soulfish5, 2021-02-17:
Hi Dave. 

I pulled down the library and it fails to build because ResistiveTouchScreen.h doesn't appear to be checked in. Can you add it? 

Thanks,

Brandon

davetcc, 2021-02-17:
Hi there the beta version needs the latest IoAbstraction 1.7.2. Can you check IoAbstraction and TaskManagerIO are up to date.

davetcc, 2021-02-17:
IoAbstraction 1.7.2 and TaskManagerIO 1.1.2. But don’t forget you’ll need to switch streams to beta in the designer UI too if you want to use the beta version. You can also use the beta 2.0 designer with 1.7 libraries.

soulfish5, 2021-02-17:
> Hi there the beta version needs the latest IoAbstraction 1.7.2. Can you check IoAbstraction and TaskManagerIO are up to date.

Ahh. I missed the IoAbstraction update. Thanks! 

Is this stuff in addition to your day job?

davetcc, 2021-02-18:
Just for the sake of others, tcMenu Designer 2.0 BETA can happily work with the 1.7 libraries, you have no need to switch stream yet, the designer is pretty much ready, the libraries are not yet ready for wide usage.

closed 2021-03-07 by davetcc