TcMenu/tcMenuLib #91: Menu Navigation

jacowestwood, 2021-02-09:
When navigating thru a menu with submenus it always jumps to the top of the main menu when exiting a submenu.
Is there a way to return to the old position in the main menu and continue navigating without having to start from the top?

davetcc, 2021-03-10:
This is in progress now, it will keep up to 4 levels of history, so allowing four menu levels (EG RootMenu -> Settings -> Advanced -> ExtraAdvanced), and a dialog. That's about as deep as the non-recursive iterator can go much deeper without defining redefining MAX_MENU_DEPTH anyway.

davetcc, 2021-03-10:
Tested and committed to master ( mbed and ESP32 ). Will be in V2.0.

davetcc, 2021-04-02:
There's a few edge cases around lists where the new navigation is not working, other than that, all is working.

davetcc, 2021-04-02:
Working for all tested cases in the 2.0 beta.

closed 2021-04-02 by davetcc