TcMenu/tcMenuLib #98: Multiple foward declarations of menuItem-callbacks when reusing the same callback in different menuItems

fs63, 2021-03-17:
Hi Dave,

I've tried to re-use an already existing menuItem-callback in several menuItems. The automatic code creation produces repeats of the same callback forward declaration in XYZ_menu.h which of course leads to a complier error. In XYZ.ino (XYZ.cpp) yet the callback is correctly inserted only once.

Commenting out / deleting the superfluous declararions is not big issue, but a little bit annoying when dealing with large menus and many repeats.

Thanks for your help, Frank

davetcc, 2021-03-17:
Yes agreed, it should not repeat the same declaration many times. Could you tell me which designer you are running from the about box?  I'll add this to the unit tests so that it gets fixed in the next version.

fs63, 2021-03-17:
Yes, of course. I'm running TcMenu Designer 1.7.13, downloaded from MS. Project devlopement for esp32.
Frank

davetcc, 2021-03-17:
I managed to recreate this and add a unit test around it. It no longer duplicates the forward declaration on the development branch. I'll be doing another beta for all platforms in the next week or so. A  store release will follow a little after.

davetcc, 2021-04-02:
This should be fixed in tcMenu 2.0, it is also added to the designer automated unit tests for both Xamarin and Java versions.

davetcc, 2021-04-02:
It is in the current beta version

closed 2021-04-02 by davetcc