TcMenu/tcMenuLib #118: Stm32 bluepill compiling problem

TheSnowFall, 2021-07-29:
having error like this

`In file included from C:\Users\Erfan Mahmud\Documents\Arduino\first_menu\first_menu.ino:1:0:
first_menu_menu.h:26:8: error: 'iotouch' does not name a type
 extern iotouch::ResistiveTouchInterrogator touchInterrogator;
        ^
first_menu_menu.h:27:8: error: 'MenuTouchScreenManager' does not name a type
 extern MenuTouchScreenManager touchScreen;
        ^
In file included from C:\Users\Erfan Mahmud\Documents\Arduino\first_menu\first_menu_menu.cpp:12:0:
first_menu_menu.h:26:8: error: 'iotouch' does not name a type
 extern iotouch::ResistiveTouchInterrogator touchInterrogator;
        ^
first_menu_menu.h:27:8: error: 'MenuTouchScreenManager' does not name a type
 extern MenuTouchScreenManager touchScreen;
        ^
first_menu_menu.cpp:21:1: error: 'iotouch' does not name a type
 iotouch::ResistiveTouchInterrogator touchInterrogator(A2, B8, B9, A3);
 ^
first_menu_menu.cpp:22:1: error: 'MenuTouchScreenManager' does not name a type
 MenuTouchScreenManager touchScreen(&touchInterrogator, &renderer, iotouch::TouchInterrogator::LANDSCAPE);
 ^
C:\Users\Erfan Mahmud\Documents\Arduino\first_menu\first_menu_menu.cpp: In function 'void setupMenu()':
first_menu_menu.cpp:49:5: error: 'touchScreen' was not declared in this scope
     touchScreen.start();
     ^
Multiple libraries were found for "Wire.h"
 Used: C:\Users\Erfan Mahmud\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\Wire
 Not used: C:\Users\Erfan Mahmud\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\WireSlave
Multiple libraries were found for "Adafruit_ILI9341.h"
 Used: C:\Users\Erfan Mahmud\Documents\Arduino\libraries\Adafruit_ILI9341
 Not used: C:\Users\Erfan Mahmud\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\Adafruit_ILI9341
Using library tcMenu at version 2.1.3 in folder: C:\Users\Erfan Mahmud\Documents\Arduino\libraries\tcMenu 
Using library IoAbstraction at version 2.0.3 in folder: C:\Users\Erfan Mahmud\Documents\Arduino\libraries\IoAbstraction 
Using library TaskManagerIO at version 1.2.3 in folder: C:\Users\Erfan Mahmud\Documents\Arduino\libraries\TaskManagerIO 
Using library Adafruit_GFX_Library at version 1.10.10 in folder: C:\Users\Erfan Mahmud\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit_ILI9341 at version 1.5.8 in folder: C:\Users\Erfan Mahmud\Documents\Arduino\libraries\Adafruit_ILI9341 
Using library SPI at version 1.0 in folder: C:\Users\Erfan Mahmud\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\SPI 
Using library Wire at version 1.0 in folder: C:\Users\Erfan Mahmud\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2021.5.31\libraries\Wire 
Using library Adafruit_BusIO at version 1.8.3 in folder: C:\Users\Erfan Mahmud\Documents\Arduino\libraries\Adafruit_BusIO 
exit status 1
'iotouch' does not name a type
`

any solution?

davetcc, 2021-07-30:
Seems quiite strange that it's tripping up on the namespace:

`first_menu_menu.h:26:8: error: 'iotouch' does not name a type`

Can you let me know which version of tcMenu Designer and plugin versions you are using? As the 2.1 plugins only work with at least the 2.1 designer. On the latest versions, the plugin versions are obtained from the settings dialog.

davetcc, 2021-09-01:
Closed, no response from the user, probable old version of IoAbstraction.

closed 2021-09-01 by davetcc