TcMenu/tcMenuLib #124: [Find and fix]AnalogMenuItem val>32767 Shown as negative

jujf1111, 2021-08-13:
env：SSD1306AsciiAvrI2c，Arduino nano（ATmega328p, tcMenuLib:2.1.3.
Reason: The variable type of getCurrentValue is uint16_t, which is forced to be converted to int16_t.

MenuItems.cpp line195(from: WholeAndFraction AnalogMenuItem::getWholeAndFraction() const ):
original：
`int calcVal = int16_t(getCurrentValue()) + getOffset();`
modified：
`uint16_t calcVal = uint16_t(getCurrentValue()) + getOffset();`

davetcc, 2021-08-29:
Apologies for the delay in answering this. We've been trying to prepare 2.2.

Analog menu item is designed for values that can be presented and edited with a rotary encoder, up/down buttons, or a slide bar on a touch screen. So realistically the range would normally be far less than +/-32K. We could make them 32 bit in a future release, but given their purpose how would you edit such a value?

Sometimes we need to edit values that are very large, maybe even in the billions, or with many decimal places, Editable large numbers allow for this, they support up to 12 digits with absolute accuracy. They are treated as BCD, and there are conversion helper functions.

Lastly, there are float menuitems, these are designed for larger status values that are read-only, and fall outside of the range of AnalogMenuItem, allowing for a much wider range of values, you can currently configure the number of decimal places. In a future release, we'll extend support and add an optional minimum-maximum range and 4 character unit.

jujf1111, 2021-09-07:
I use AnalogMenuItem to display the value obtained by a sensor.
I think a boundary check should be done before a value is displayed.
The variable type of getCurrentValue is uint16_t.
Should not become int16_t without checking;

davetcc, 2021-09-07:
> Should not become int16_t without checking

AnalogMenuItem can represent both positive and negative values. The offset in negative cases is a negative value. EG a decibel scale from -100 through to +100. This means the conversion to int16 is needed. It should never usually be an issue as Analog items are designed for smaller values that can be edited with a rotary encoder or slider. 

We've thought about making analog item 32bit in an upcoming release, but that comes with issues for 8-bit devices. We could also make it possible to display a larger value by checking if the offset is greater than 0 and NOT using the decimal conversion classes, but this causes complexity in the code.

The original intent is that if you need above int16 range, then you can use FloatMenuItem that has a much larger range for read-only values from a sensor. If you need an editable value larger than that range, use the large number support that edits one digit at a time.

See https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/analog-menu-item/

jujf1111, 2021-09-07:
Thank you for your reply, I have understood its original intention.
But when "getCurrentValue()>32767", the display is still incorrect.

davetcc, 2021-09-13:
> But when "getCurrentValue()>32767", the display is still incorrect.

With AnalogMenuItem, it is not possible to present a positive value > 32767. It should only be correct when presenting values within the range of -32768 and +32767. The documentation was very recently changed to make this point clearer:

https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-item-types/analog-menu-item/

davetcc, 2021-10-17:
Change the int16_t to int32_t in the code that does the display rendering. This should allow values in the range of
-32768 -> 65535.