TcMenu/tcMenuLib #132: Rotary Encorder KY-040 not working

antwal, 2021-12-31:
Hi all,

I've created four menu items with one having a two-item submenu; the problem is that the KY-040 rotary encorder works the click but the shift doesn't work; I have tried several encoders of the same type but they all have the same problem.

how can i solve? Is it a problem with the encoder?

Thanks.

davetcc, 2022-01-01:
When raising an issue, please always provide:

* Board type.
* Pins on which the faulting device connected.
* Version of IoAbstraction and tcMenuLib that are being used.

Is at least pin A on an interrupt pin? At the moment, it is mandatory that pin A be an interrupt pin.

antwal, 2022-01-01:
hi,

board: arduino uno REV3 & arduino nano v3
pins tested: 4,3,2 - 10,9,8

code generated with tcMenu 2.2.5 Mac

attach [example](https://pastebin.com/d58SV2DJ) used

Thanks.

davetcc, 2022-01-01:
Uno with tcMenu and I2C often doesn't even compile because there is not enough space for the wire library. But even if it does compile, I suspect that the board is running out of memory at runtime. 

The library only really supports Uno with the DfRobot shield on direct pins (IE no use of the I2C Wire library). If you want to use Uno, keep the sketch as small as possible, and use a DfRobot LCD shield, as that uses direct pins on the device.

antwal, 2022-01-01:
hi, 

i see on arduino ide log: 

```
Sketch uses 25894 bytes (80%) of program storage space. Maximum is 32256 bytes.
Global variables use 771 bytes (37%) of dynamic memory, leaving 1277 bytes for local variables. Maximum is 2048 bytes.
```

it working without problems on Arduino Uno

for Arduino nano the log is:

```
Sketch uses 25894 bytes (84%) of program storage space. Maximum is 30720 bytes.
Global variables use 771 bytes (37%) of dynamic memory, leaving 1277 bytes for local variables. Maximum is 2048 bytes.
```

and also it working without problems.

not working the rotary encoder, in this moment.

Thanks.

davetcc, 2022-01-01:
Ack, I'm always suspicious on Uno boards as it's right on the edge of what the library can support. What's reported at compile time is not the whole story as there are a few one-off allocations during setup.

What I would suggest is try using one of the IoAbstraction examples, and see if that works. TcMenu uses IoAbstraction to provide the encoder support, so if that examples works/doesn't we can determine where the problem is.

I suggest 'directionOnlyEncoder' as the simplest example, change the pins as appropriate and see if that works, it should log to the serial port.

https://github.com/davetcc/IoAbstraction/blob/master/examples/directionOnlyEncoder/directionOnlyEncoder.ino

antwal, 2022-01-02:
Hi,

not working, the log is: 

```
Encoder change 
Encoder change 
Button pressed Pressed
Button pressed Pressed
Button pressed Pressed
```

Thanks.

antwal, 2022-01-02:
hi,

according to the [documentation](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/) the arduino uno supports interrupts only on pins 2, 3; instead the code you sent me does not use these pins

After making the necessary changes to the pins, I did the test again and now it works

serial log:
```
Encoder change 
Encoder change down
Encoder change down
Encoder change down
Encoder change down
Encoder change down
Encoder change up
Encoder change up
Encoder change up
Encoder change up
```

Thanks,

davetcc, 2022-01-08:
Looking at this, it is probable that either there is a misconfiguration somewhere in the menu (not likely looking at your sketch), or the Uno is just running out of memory. We have never tried using Uno with an I2C display that needs wire library. Normally, we only use Uno with an LCD shield that uses direct pins on the Uno.

If you zip up the faulty example I'll see if I can run it on my Uno, but it is likely to crash there too, and determining if it ran out of memory is very hard as AVRs just tend to reset when that happens. They are not like ESP or larger boards where a full diagnostic is available.

Really, given the size of the Uno (32K FLASH, 2K RAM) it is surprising that you can run something as complete as tcMenu on it at all!

davetcc, 2022-02-13:
Closing for now, as I said earlier we have never tried I2C with tcMenu on Uno, and suspect it would require too much memory

closed 2022-02-13 by davetcc