TcMenu/tcMenuLib #139: Add Exit-Callback for SubMenuItems

emptynick, 2022-04-20:
Hello,

I'm getting better and better in working with your library - thanks for that.

What I am implementing right now is a submenu in which you can set the IP, gateway and subnet of the ethernet adapter.
This is what my structure looks like:

- Ethernet Settings (Submenu)
    - Back (from submenu)
    - IP Address (IP)
    - Subnet mask (IP)
    - Gateway (IP)
    - Apply (Action)
- System
    - ....

Once you have changed everything to your liking, you'd hit an action-item to apply the changes (and restart ethernet).

Now, whenever you just go back to the main menu through the back-button or when the menu resets-to-default, I'd like to restore the current values to all those items (eg setting the IP address to its actual value).
For that it would be really great to have a callback that gets called whenever a submenu is "exited".

davetcc, 2022-04-23:
This is a small enough to go into the 2.3 release coming up soon. There is already a `MenuManagerObserver` that you added to menuMgr to be notified of changes, it has a method for edit start and edit end. It already gets calls when a new menu is displayed. I'll make the edit end call as follows:

* Menu goes onto display - menuEditStarting(..) and only display if returns true
* Menu is being replaced - menuEditStarting(..) - if returns true the menu is replaced
* Menu has been replaced - call menuEditEnded(..) as the item has been removed from display.

Make sure you never disallow ROOT from being displayed in this observer, this could have unpredictable results.

davetcc, 2022-04-23:
You can see a menu example that already implements this observer:

https://github.com/davetcc/tcMenuLib/tree/master/examples/dynamicMenuItems

davetcc, 2022-04-23:
It seems odd that the end editing call will come after the new edit notification, but this is because you can prevent the new edit call by returning false, and in that case, you would not move to a new menu.

closed 2022-05-07 by davetcc