TcMenu/tcMenuLib #146: Due build failure

NachtRaveVL, 2022-08-13:
Found this while attempting to compile 2.3.1 for Due.

Not a platform I intend on actually using I just happened to have one laying around and decided to give it a whirl, make sure it's not dead. Figured you'd appreciate knowing.

```
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp: In member function 'void tccore::MenuItemDelegate::setReadOnly(bool)':
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:25:11: error: invalid user-defined conversion from 'tccore::MenuItemDelegate::setReadOnly(bool)::__lambda0' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}' [-fpermissive]
}, ALL);
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:22:43: note: candidate is: tccore::MenuItemDelegate::setReadOnly(bool)::__lambda0::operator int (*)(MenuItem*, bool)() const <near match>
onEachItem([](MenuItem* item, bool flg) {
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:22:43: note:   no known conversion for implicit 'this' parameter from 'int (*)(MenuItem*, bool)' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}'
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp: In member function 'void tccore::MenuItemDelegate::setLocalOnly(bool)':
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:33:15: error: invalid user-defined conversion from 'tccore::MenuItemDelegate::setLocalOnly(bool)::__lambda1' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}' [-fpermissive]
}, ALL);
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:30:43: note: candidate is: tccore::MenuItemDelegate::setLocalOnly(bool)::__lambda1::operator int (*)(MenuItem*, bool)() const <near match>
onEachItem([](MenuItem* item, bool flg) {
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:30:43: note:   no known conversion for implicit 'this' parameter from 'int (*)(MenuItem*, bool)' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}'
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp: In member function 'void tccore::MenuItemDelegate::setVisible(bool)':
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:41:15: error: invalid user-defined conversion from 'tccore::MenuItemDelegate::setVisible(bool)::__lambda2' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}' [-fpermissive]
}, ALL);
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:38:43: note: candidate is: tccore::MenuItemDelegate::setVisible(bool)::__lambda2::operator int (*)(MenuItem*, bool)() const <near match>
onEachItem([](MenuItem* item, bool flg) {
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:38:43: note:   no known conversion for implicit 'this' parameter from 'int (*)(MenuItem*, bool)' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}'
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp: In member function 'void tccore::MenuItemDelegate::setChangedAndRemoteSend()':
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:49:15: error: invalid user-defined conversion from 'tccore::MenuItemDelegate::setChangedAndRemoteSend()::__lambda3' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}' [-fpermissive]
}, ALL);
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:45:43: note: candidate is: tccore::MenuItemDelegate::setChangedAndRemoteSend()::__lambda3::operator int (*)(MenuItem*, bool)() const <near match>
onEachItem([](MenuItem* item, bool flg) {
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:45:43: note:   no known conversion for implicit 'this' parameter from 'int (*)(MenuItem*, bool)' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}'
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp: In member function 'void tccore::MenuItemDelegate::setChangedOnly()':
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:56:15: error: invalid user-defined conversion from 'tccore::MenuItemDelegate::setChangedOnly()::__lambda4' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}' [-fpermissive]
}, ALL);
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:53:43: note: candidate is: tccore::MenuItemDelegate::setChangedOnly()::__lambda4::operator int (*)(MenuItem*, bool)() const <near match>
onEachItem([](MenuItem* item, bool flg) {
^
C:\Users\name\Documents\Arduino\libraries\tcMenu\src\extras\MenuItemDelegate.cpp:53:43: note:   no known conversion for implicit 'this' parameter from 'int (*)(MenuItem*, bool)' to 'tccore::MenuItemDelegate::ItemDelegateFn {aka bool (*)(MenuItem*, bool)}'
exit status 1
```

davetcc, 2022-08-13:
Yes, it's great to know, thanks a lot. I try it on quite a few boards, the ones that most people seem to target often. I don't have a due hanging around but I'll just set that to be the board type and give it a try, I can at least make it compile.

Thanks, again for helping out!

davetcc, 2022-10-17:
Found it, the way `true`, `false` and `bool` are defined is a bit off, at least it compiles now.

closed 2022-10-25 by davetcc