TcMenu/tcMenuLib #153: SSD1306Ascii - In function `main': undefined reference to `Display variable name'

jokubasver, 2022-09-29:
I can't get the SSD1306Ascii display to work - it fails at the linker and simply gives an error "undefined reference to '_Display variable name_'"

> Linking .pio\build\uno\firmware.elf
C:\Users\USER\AppData\Local\Temp\ccHhzjXq.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x4e2): undefined reference to `gfx'
<artificial>:(.text.startup+0x4e4): undefined reference to `gfx'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\uno\firmware.elf] Error 1

Using an Arduino Uno, tried with Arduino IDE and PlatformIO.
Tried changing the display variable name to oled, same thing.
Tried SSD1306AsciiAvrI2c and SSD1306AsciiWire, same thing.

Is the SSD1306Ascii library still supported?

davetcc, 2022-09-29:
I'll make it clearer in the generator text, but you need to declare a variable of that type in your sketch. IIRC there were a few different ways the display could be created, so you declare the variable in your sketch and initialise it in the sketch file before calling setupMenu.

You'd normally declare the variable (same name as you put here) as a global in your INO file, and call begin on it early on in your sketch.

davetcc, 2022-10-13:
Add a few basic simple cases to declare the variable on behalf of the user.

davetcc, 2022-10-15:
See https://github.com/davetcc/tcMenu/issues/230

closed 2022-10-17 by davetcc