TcMenu/tcMenuLib #167: Is there a way to save single MenuItem's value to EEPROM ?

luwenjin, 2022-11-07:
I have a bunch of settings  under `Settings` Menu, and I also have some standalone options under other menus.

These stand alone options need to be saved when changed.
But currently, it seems I can only save the whole menu data to EEPROM.

It would be nice if we can save single MenuItem's value to EEPROM.

davetcc, 2022-11-07:
For questions like this should we start using the github discussions? Something that turns out to be an issue can easily be converted into an issue from there.

luwenjin, 2022-11-07:
Sorry, I was worrying that if I put all (UI&lib) issues in the UI's repo, it would make your git less organized.
Since you like it that way, I'm also glad to post issues there for convenience 😄 

I'll move this issue there.

davetcc, 2022-11-12:
I'll reopen it to track the work.

davetcc, 2022-11-12:
See https://github.com/davetcc/tcMenu/issues/250 for the main issue

davetcc, 2022-11-13:
https://www.thecoderscorner.com/products/arduino-libraries/tc-menu/menu-eeprom-integrations/#loading-or-saving-a-single-menu-item

closed 2022-12-17 by davetcc