TcMenu/tcMenuLib #184: TcMenu 3.1.0 not released in PlatformIO registry

vzahradnik, 2023-02-14:
After TcMenu Designer 3.1.0 was released, I'm not able to compile my project. It requires TcMenu library 3.1.0 but it wasn't released in PlatformIO registry.

[image]

davetcc, 2023-02-14:
Apologies released it just now. Along with IoAbstraction 3.1.0

davetcc, 2023-02-15:
Please let me know if that fixed the issue.

I think for future releases I need to have a runbook in excel or similar, so that I can keep track of all the steps, of which there are many.

davetcc, 2023-02-15:
Update is that I have spun up an AWS Ubuntu image and started to install the build server dependencies, so soon I will have a production build of 3.1 ready

davetcc, 2023-02-16:
I’ll keep open until the Linux package is built.

davetcc, 2023-02-17:
Closing out now, if there are any residual actions left to be done, please reopen.

closed 2023-02-17 by davetcc