TcMenu/tcMenuLib #206: Set up automated PlatformIO releases

vzahradnik, 2023-11-12:
**Is your feature request related to a problem? Please describe.**
PlatformIO registry requires the maintainer to manually push the updates after each release. The release process is different from Arduino where user has just to update the manifest.

**Describe the solution you'd like**
We can use GitHub Actions to automate the process. After a new release is made or a new tag is added to the main branch, GitHub Actions will push the changes to PlatformIO registry.

vzahradnik, 2023-11-21:
@davetcc please provide the list of boards you support (if you have any). Also, if you provide the procedure to run your unit tests, I would like to automate tests as well.

davetcc, 2023-11-21:
Hi there, as per the other day, the most up-to-date list of boards that are known working is here:

https://www.thecoderscorner.com/products/arduino-libraries/#defining-build-flags-and-platform-settings

The tests just run in exactly the same way as Arduino sketches (IE if you were to load one of those test projects into Arduino, or change the source directory to the test folder (platformIO) then you could run them on device.

vzahradnik, 2024-07-05:
Implemented in #219. Since examples have many dependencies, for now I included support for ESP and AVR examples. Remaining examples can be added later.

closed 2024-07-05 by vzahradnik