TcMenu/tcMenuLib #218: Code won't compile with u8g2

nmeurer, 2024-06-20:
Hi there,
after searching for quite some time, I still can't find the issue. My main strength also isn't in programming so any help would be very much appreciated.

**Describe the bug**
A clear and concise description of what the bug is.
I can't get the generated code working. I modified nothing and just made changes in the editor.

**Confirm that the bug is in the embedded library**
I interpret the given error as a bug in the library, but I am by no means an expert.

**To Reproduce**
Please include a simple sketch that recreates the problem unless the problem is so easily recreated that such a sketch is not needed.
Create any menu using the u8g2 quickstart with these settings:

[image]

 
**Expected behavior**
I expected the code to compile

**Screenshots**
If applicable, add screenshots to help explain your problem.
Compiler error message:
[image]

**Device Specs**
An arduino mega with a ST7920 128x64 attatched via hardware SPI

**Additional context**
I also tried using the manual u8g2 setup, but I don't seem to get it right. I'd love to see an example of how it is supposed to be done.

Regards

vzahradnik, 2024-07-05:
Hmm... It works for me. Check the contents of `*_menu.h`. It should contain a variable like this:

`extern U8G2_SSD1309_128X64_NONAME0_F_HW_I2C gfx;`

Then check `*_menu.cpp` whether you can see an initialization code.

`U8G2_SSD1309_128X64_NONAME0_F_HW_I2C gfx(/*some parameters*/);`

Third, make sure `*_menu.h` is included withing `*_menu.cpp`.

davetcc, 2024-09-08:
Hi we've had no update on this in several months, we'll close the issue for now. Should you still be having the same issue please reopen and provide the project zipped up.

closed 2024-09-08 by davetcc