TcMenu/tcMenuLib #241: G'day. Any chance of getting a plugin for mcefriend dispplays

Moorey79, 2025-04-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Yes _(well to me anyhow)_  The problem is there isnt a ready to setup for the McuFriend library's, id7575 (and i'm too stupid to be able to get the custom one working)**

**Confirm that the bug is in the embedded library**
If the bug is in the designer, or you are not sure, raise in the main [tcMenu repo](https://github.com/TcMenu/tcMenu).

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Some one to come and write my code for me. (failing that, a new plugin??)

**Additional context**
Add any other context or screenshots about the feature request here.
Not really.

vzahradnik, 2025-04-29:
Hi @Moorey79 currently we are seriously understaffed. We're focusing on bugfixes and improvements in the overall TcMenu experience. We might consider making such plugin in the future but not in the near future. Perhaps if you provide more information, I can guide you.

Also it will help if you provide information about the board you are using, etc.

vzahradnik, 2025-04-29:
I guess this is the GFX library for the display?
https://github.com/prenticedavid/MCUFRIEND_kbv

Moorey79, 2025-04-30:
Yeah thats the one.

Thank you for getting back to me.
I completely understand and if i ever get around to making one, i'll pass
it on.

thanks.
Steve

davetcc, 2025-11-02:
Adding plugins for displays got a bit easier recently, if you can provide the repo to the display you're using so I can look at the examples and work out how to configure it, 

Then I can try and add a plugin for it, but I cannot test it as I don't have one of those displays.